microsoft/Windows-driver-samples #1068: While i build the Sample "Windows-driver-samples\audio\simpleaudiosample" getting Error "The build tools for WindowsKernelModeDriver10.0 Can not be found,to build using the WindowsKernelModeDriver 10.0 build tools,please install WindowsKernelModeDriver 10.0 build toold, Alternatively, you may upgrade to the current visual studio tools by selecting the Project menu or right -click the solution "Retarget solution"."

HBTest13, 2024-01-08:
While i build the Sample **"Windows-driver-samples\audio\simpleaudiosample"** i am getting Error "The build tools for WindowsKernelModeDriver10.0 Can not be found, to build using the WindowsKernelModeDriver 10.0 build tools ,Please install WindowsKernelModeDriver 10.0 build tool, Alternatively, you may upgrade to the current visual studio tools by selecting the Project menu or right -click the solution "Retarget solution"."

I have checked all the possible ways like checked in Win11, update WDK and SDK with compatible version.

And currently i am using VS 2022 and Win11 to build this project.

Can you please help out to solve this error.

hhhheying, 2024-04-09:
Hello,

I encountered a similar issue and was able to resolve it. Here are some steps you might find helpful:

1. Verify SDK and WDK Versions: Ensure that you have the correct versions of the Windows SDK and Windows Driver Kit (WDK) installed. Compatibility with your version of Visual Studio is crucial.

2. Integration with Visual Studio: If you haven't already, integrate the Windows SDK and WDK with Visual Studio:

- Navigate to the directory where the Windows Kits are installed, typically located at C:\Program Files (x86)\Windows Kits\10\.
- Look for the Vsix  within the Windows Kits directory.
- Find the file named WDK.vsix and double-click to run the installer. Make sure to close Visual Studio before doing this to avoid any conflicts.

I hope this helps!

NeoAdonis, 2024-06-04:
@HBTest13: It seems like an issue with your WDK installation. Following the steps hhhheying suggested might help you with it.

I'm closing the issue for now but please let me know if the issue was resolved or if you encountered more problems.

closed 2024-06-04 by NeoAdonis

tikonen, 2024-07-19:
Putting this here In case somebody googles this problem. My issue was that somehow the target had changed to ARM when driver kit was installed for x64.
[image]

ddbb2017, 2025-01-26:
@hhhheying how did you determine which version of WDK is needed for the solution? I can see what is installed but that message alone "_The build tools for WindowsKernelModeDriver10.0 Can not be found_" doesn't tell you which version it is missing.

hhhheying, 2025-02-04:
@ddbb2017  I am using WDK version 10.0.26100.1, and I believe that using the latest version of WDK should also work for running the example code. Hope this helps!