microsoft/Windows-driver-samples #1077: AVSSAMP example  not working

shiqifeng2000, 2024-01-18:
hi, thanks for these samples, I would like to create a virtual camera with both video & audio, and perhaps some decoding functions

avshws working for me but got no sound, then I try avssamp. it builds successfuly, but not working.

I am using win11, got windows driver kit 10 installed, and have the cer file registered to root
after that i tried:
1. using hdwwiz, after choosing inf file, got option to continue
[image]

2. try using devcon to register, it succeed, but unable to play
[image]

[image]

i created a avssamp.wav with my ffmpeg using pcm16le format, 1 channel, 44100hz, im sure it's valid
[image]

could you let me know if anything I config wrong

ColorfulDick, 2024-06-11:
我也遇到了这个问题，驱动成功编译并安装，也能进行测试，测试时每次显示的音量都是固定的百分之五十，而不是像正常的音频设备驱动那样是会跳动的音量；测试时听不到任何声音

shiqifeng2000, 2024-06-13:
> 我也遇到了这个问题，驱动成功编译并安装，也能进行测试，测试时每次显示的音量都是固定的百分之五十，而不是像正常的音频设备驱动那样是会跳动的音量；测试时听不到任何声音

avshws好用，avssamp就不行，很奇怪，filter模式就是不行嘛，是版本问题嘛，也没有官方人来回答下 :(

shiqifeng2000, 2024-06-18:
> > t
> 
> 老哥，avssamp是摄像头的吗？我本来找了个模型想实现一个基于虚拟摄像头实时更换背景的功能，编译安装了avscamera以后，在“设备管理器”里面也没有看到相应的设备；不知道是缺了哪一步

avssamp是虚拟音频，但是也有视频的pin，有3个分类KSCATEGORY_VIDEO,KSCATEGORY_CAPTURE, KSCATEGORY_AUDIO。而avshws是虚拟视频，有3个分类KSCATEGORY_VIDEO,KSCATEGORY_CAPTURE,KSCATEGORY_VIDEO_CAMERA

v-junyli, 2024-11-18:
@microsoft/cameraplatdev

walker-WSH, 2025-05-20:
@shiqifeng2000   你好 请问你最后跑通了avssamp吗？

我搜索到一个其他人分析的文档 倒是可以正常在graphEdit的WDM capture devices下找到amssamp了  但依旧不能被zoom或obs识别到设备：
https://www.reddit.com/r/learnprogramming/comments/y2lj1l/installing_avssamp_virtual_camera_driver_in/