microsoft/Windows-driver-samples #1100: PC got Blue Screen of Death after installing SimpleAudioSample

maruf-rahad, 2024-02-05:
I recently attempted to install the driver SimpleAudioSample.inf by devcon obtained from repository on windows 11.

After completion of the installation process, my system encountered a crash and leading to an automatic restart with **_Blue Screen of Death_**. I cannot figure it out the reason I am facing this as I am new to kernel mode driver development. I am attaching the picture of it for better understanding. 

Please suggest me a way of avoiding this or the root cause of it for further analysis. Thanks.

[image]

maruf-rahad, 2024-02-09:
I have figured it out. It's happening only for debug build version.
In debug version the driver will print all logs including error logs which will call INT3 (Interrupt 3 - trap to debugger).
This will create BSOD(Blue Screen of Death)
 If you use the release build version then you may not face this problem.
 Thanks

N00ree, 2024-04-14:
+1 Thanks for providing the solution to this. (Also a newbie learning this ^^) 

Besides the fact that the debug version didn't work, I somehow destroyed my VM and couldn't install any drivers anymore. I also used "devcon remove" to remove all the device but this didn't help. God bless that I had a snapshot of my VM (Win 11 on Hyper-V). After reverting and starting over it all worked.