microsoft/Windows-driver-samples #1105: Cleanup Branch names

rodwiddowson, 2024-02-12:
Looks like you have moved to the new branch naming (`origin/main`) but you have left the old branch (`origin/master`)  around.

My local main branch was pointing at master so I accidentally missed 9 months of changes.

I _think_ that most people would  be happier to see their repo break than go out of date silently. Could you do something about this sometime?

Thanks

NeoAdonis, 2024-04-24:
Hi! First, I'm sorry for this late reply.

The change move was done according to [GitHub's instructions](https://github.com/github/renaming). Repo owners are informed of it when they access the repo site or try to push into the old branch name (in this case, `origin/master`). It seems that you weren't notified about this when trying to pull new changes, though. I'm sorry to know that something similar happened to you and might have happen to other people as well.

I'll review our current documentation to make this behavior clearer. Thanks for letting us know!

v-junyli, 2024-12-09:
looks like the old master branch has been deleted. Closed.

closed 2024-12-09 by v-junyli