microsoft/Windows-driver-samples #1106: FASTFAT Errors on Surprise removal

rodwiddowson, 2024-02-12:
I am debugging and issue with my own file system driver and IRP_MN_SURPRISE_REMOVAL so I built a debug version of FASTFAT and started testing.

In my initial set of tests I got a series of PAGED_CODE asserts in the classpnp completion routine  Because my own driver was loaded I didn't take any notes but tried to reproduce without it.

Investigations are ongoing but so far I have seen this (FATDATA.c - 478)
```
 if (ExceptionCode == STATUS_VERIFY_REQUIRED) {

            PDEVICE_OBJECT Device = NULL;

            DebugTrace(0, Dbg, "Perform Verify Operation\n", 0);

            //
            //  Now we are at the top level file system entry point.
            //
            //  Grab the device to verify from the thread local storage
            //  and stick it in the information field for transportation
            //  to the fsp.  We also clear the field at this time.
            //

            Device = IoGetDeviceToVerify( Irp->Tail.Overlay.Thread );
            IoSetDeviceToVerify( Irp->Tail.Overlay.Thread, NULL );

            if ( Device == NULL ) {

                Device = IoGetDeviceToVerify( PsGetCurrentThread() );
                IoSetDeviceToVerify( PsGetCurrentThread(), NULL );

                NT_ASSERT( Device != NULL );
            }
```

 I'll keep on testing (with and without my drivers) and I'll fill in more details as I find them.

darwin-msft, 2024-06-04:
STATUS_VERIFY_REQUIRED is expected. It is returned whenever fastfat needs to validate that the underlying storage is as expected. Usually after a media ejection, or power state transition or something like that.

v-junyli, 2025-11-08:
test stale label.