microsoft/Windows-driver-samples #1146: WMI ACPI object conversion rules

Wer-Wolf, 2024-03-25:
The example WMI ACPI code contains no information on how ACPI objects are serialized into the WMI buffer.
According to [this article](https://learn.microsoft.com/en-us/windows-hardware/drivers/kernel/driver-defined-wmi-data-items), driver operate on plain WMI buffers for exchanging data.

So far i found out that:

- ACPI integers seem to be converted into 32 bit WMI integers (even if they are 64 bit long)
- ACPI buffers are copied as-is
- ACPI strings are converted into WMI strings

What i do not know is how ACPI Packages are processed and if the ACPI-WMI mapper takes the alignment of the WMI data items into account. Would it be possible to write a short guide about ACPI object conversion rules for firmware developers?

Thanks,
Armin Wolf

Wer-Wolf, 2024-06-21:
Any progress?

Wer-Wolf, 2024-06-22:
I did some research myself, and i found found at that ACPI packages are "unpacked" by converting the data items inside with alignment:

- integers: 4 byte
- strings: 2 byte
- buffers: 1 byte

Wer-Wolf, 2024-06-22:
I also found out that there is a bug inside the wmiacpi.sys driver which can be triggered by returning an ACPI package containing another ACPI package from an ACPI method.

**ASL-Code:**

    Return (Package () {
        0xabcd,
        Buffer () { 0x1 },
        Package () {
                0xfeed,
                "TEST"
        },
        Buffer () { 0x2 },
        0xabcd
    }

**Result:**

    cd ab 00 00 // int
    01 // Buffer
    00 00 // Type 0 (ACPI_METHOD_ARGUMENT_INTEGER)
    04 00 // 4 bytes
    ed fe 00 00 // int data
    01 00 // Type (ACPI_METHOD_ARGUMENT_STRING)
    05 00 // 5 bytes
    54 45 53 54 00 // string data "TEST"
    ...

Basically the driver starts to copy raw `_ACPI_METHOD_ARGUMENT_V1` structs when encountering a ACPI package, treating it like a ACPI buffer.

Trying to recursively parse such ACPI packages however will allow the firmware to trigger stack overflows, so the best solution would be to just return an error.

Wer-Wolf, 2024-06-22:
Also it seems that there exists a buffer overflow somewhere inside the WMI stack, since when defining an WMI event with a single `uint32` payload, the WMI stack somehow accepts ACPI buffers containing only two bytes of data.

The upper two bytes of the resulting `uint32` are filled with garbage data, but the lower two bytes contain the data from the buffer.

yinghany, 2024-06-22:
Thanks Armin. I’m currently on leave and adding @Tomas Lugo ***@***.***> and @Ria ***@***.***> to help investigate.

Thanks
Yinghan

You are receiving this email because you were assigned.

tomas, 2024-06-24:
@yinghany Please don't mention me again in this repo

Wer-Wolf, 2024-07-06:
Any progress so far?

Wer-Wolf, 2024-08-31:
It also seems that when converting UTF16 WMI strings to ACPI strings, the driver does not return an error when receiving non-ASCII characters, instead the string is modified (ö -> o, etc)

This might cause big issues if firmware developers are not aware of that.

Wer-Wolf, 2025-02-07:
Any updates on this?