microsoft/Windows-driver-samples #1170: [avstream/sampledevicemft] The face detection rectangle is missing on Windows Camera App

ispaddy, 2024-05-22:
Hi , there

If I enable `MF_DEVICEMFT_DECODING_MEDIATYPE_NEEDED` in `/avstream/sampledevicemft/stdafx.h`, the face detection blue rectangle disappears in the Windows Camera App. Can anyone provide me with some advice? Many thanks :)

```c
// @@@@ README Please check / uncheck various hash defines to enable/ disable features
// MF_DEVICEMFT_ASYNCPIN_NEEDED will show how to use Asynchronous queues
// MF_DEVICEMFT_DECODING_MEDIATYPE_NEEDED will show how to decode a compressed mediatype.. it supports H264 and MJPG
// MF_DEVICEMFT_SET_SPHERICAL_ATTRIBUTES will set the spherical attributes
// MF_DEVICEMFT_ENUM_HW_DECODERS will make the device MFT enumerate hardware decoders
// 
//#define MF_DEVICEMFT_ADD_GRAYSCALER_ 1  
#define MF_DEVICEMFT_ASYNCPIN_NEEDED           1
#define MF_DEVICEMFT_DECODING_MEDIATYPE_NEEDED 1       // <------------ enable this
//#define MF_DEVICEMFT_SET_SPHERICAL_ATTRIBUTES  1
//#define MF_DEVICEMFT_ENUM_HW_DECODERS        1

/*Some important notes regarding Device MFT
Device MFT is an asynchronous transform that is loaded in the device source
There are a few key operations that need a special mention
1) InitializeTransform: This is where the device MFT should be initialized
2) ProcessInput is where the samples are supplied to the Device MFT
3) METransformHaveOutput is how deviceMFT notifies the Device Transform Manager (DTM) in the OS to pick up samples in its queues
4) SetOutPutStreamState is where the state transition on a Device MFT will come in
*/

```

[image]

AbhinavMalviya58, 2024-05-22:
in my window  i am click then blue rectangle disappears otherwise always on 

what do you want to do so that the blue rectangle remains after clicking ?

ispaddy, 2024-05-23:
Hi @AbhinavMalviya58 , blue rectangle for face dection is only disapear when I enabled `MF_DEVICEMFT_DECODING_MEDIATYPE_NEEDED`

ispaddy, 2024-06-18:
Found a workaround to fix this issue, FYI

```diff
@@ -664,7 +664,7 @@ public:
 #endif
         )
     {
-        SetUINT32(MF_SD_VIDEO_SPHERICAL, TRUE);
+        //SetUINT32(MF_SD_VIDEO_SPHERICAL, TRUE);
     }
     STDMETHOD(AddMediaType)(
         _Inout_ DWORD *pos,

```

closed 2024-06-18 by ispaddy