microsoft/Windows-driver-samples #1177: Unable to build bthecho

steam3d, 2024-06-06:
When build the bthecho sample as described in README I get error
```
'BthEchoSampleSrv' is reserved
'BthEchoSampleCli' is reserved
```

[image]

# BthEchoSampleCli.inx file:
 
https://github.com/microsoft/Windows-driver-samples/blob/0038767e5a3f27d464d20f2d4b97f607e58aac5d/bluetooth/bthecho/bthcli/sys/BthEchoSampleCli.inx#L56

https://github.com/microsoft/Windows-driver-samples/blob/0038767e5a3f27d464d20f2d4b97f607e58aac5d/bluetooth/bthecho/bthcli/sys/BthEchoSampleCli.inx#L67

# BthEchoSampleSvr.inx file:

https://github.com/microsoft/Windows-driver-samples/blob/0038767e5a3f27d464d20f2d4b97f607e58aac5d/bluetooth/bthecho/bthsrv/sys/BthEchoSampleSrv.inx#L56

https://github.com/microsoft/Windows-driver-samples/blob/0038767e5a3f27d464d20f2d4b97f607e58aac5d/bluetooth/bthecho/bthsrv/sys/BthEchoSampleSrv.inx#L67

NeoAdonis, 2024-06-18:
This is by design: we expect samples to be changed and used only as the basis to build new drivers, rather than being built as they are.

There are ways to exclude this error. We use those, for example, to validate that the sample driver builds as expected. However, the right way to resolve this issue is to rename the service.

I'm closing this for now but feel free to add more comments if you have any other questions or issues!

steam3d, 2024-06-18:
@NeoAdonis I quote README file:
[image]

I expect to get a successful build according to the REAMDE file, but not an error. This should definitely be fixed so as not to confuse developers and generate additional issues

NeoAdonis, 2024-06-18:
@steam3d I agree. Thanks for letting me know! I'll reach out to the team to see what's the best approach to document this.