microsoft/Windows-driver-samples #1183: Missing .inf files on my host computer

vrinda41198, 2024-06-12:
I was able to successfully build my sysvad solution, but I do not see any generated .cer or .inf files. This is my build output:
`Build started at 18:40...
1>------ Build started: Project: AecAPO, Configuration: Debug x64 ------
2>------ Build started: Project: KwsAPO, Configuration: Debug x64 ------
3>------ Build started: Project: SwapAPO, Configuration: Debug x64 ------
4>------ Build started: Project: DelayAPO, Configuration: Debug x64 ------
5>------ Build started: Project: EndpointsCommon (EndpointsCommon\EndpointsCommon), Configuration: Debug x64 ------
6>------ Build started: Project: KeywordDetectorContosoAdapter, Configuration: Debug x64 ------
1>64 bit Processing .\AecApoDll.idl
1>AecApoDll.idl
5>a2dphpminwavert.cpp
5>a2dphpspeakertopo.cpp
1>64 bit Processing C:\Program Files (x86)\Windows Kits\10\Include\10.0.26100.0\um\oaidl.idl
5>a2dphptopo.cpp
3>64 bit Processing .\SwapAPODll.idl
1>oaidl.idl
1>64 bit Processing C:\Program Files (x86)\Windows Kits\10\Include\10.0.26100.0\um\objidl.idl
5>AudioModuleHelper.cpp
3>SwapAPODll.idl
1>objidl.idl
2>64 bit Processing .\KWSApoDll.idl
3>64 bit Processing C:\Program Files (x86)\Windows Kits\10\Include\10.0.26100.0\um\oaidl.idl
3>oaidl.idl
1>64 bit Processing C:\Program Files (x86)\Windows Kits\10\Include\10.0.26100.0\um\unknwn.idl
2>KWSApoDll.idl
2>64 bit Processing C:\Program Files (x86)\Windows Kits\10\Include\10.0.26100.0\um\oaidl.idl
3>64 bit Processing C:\Program Files (x86)\Windows Kits\10\Include\10.0.26100.0\um\objidl.idl
3>objidl.idl
1>unknwn.idl
1>64 bit Processing C:\Program Files (x86)\Windows Kits\10\Include\10.0.26100.0\shared\wtypes.idl
4>64 bit Processing .\DelayAPODll.idl
2>oaidl.idl
3>64 bit Processing C:\Program Files (x86)\Windows Kits\10\Include\10.0.26100.0\um\unknwn.idl
3>unknwn.idl
1>wtypes.idl
1>64 bit Processing C:\Program Files (x86)\Windows Kits\10\Include\10.0.26100.0\shared\wtypesbase.idl
4>DelayAPODll.idl
4>64 bit Processing C:\Program Files (x86)\Windows Kits\10\Include\10.0.26100.0\um\oaidl.idl
2>64 bit Processing C:\Program Files (x86)\Windows Kits\10\Include\10.0.26100.0\um\objidl.idl
2>objidl.idl
3>64 bit Processing C:\Program Files (x86)\Windows Kits\10\Include\10.0.26100.0\shared\wtypes.idl
3>wtypes.idl
4>oaidl.idl
4>64 bit Processing C:\Program Files (x86)\Windows Kits\10\Include\10.0.26100.0\um\objidl.idl
1>wtypesbase.idl
2>64 bit Processing C:\Program Files (x86)\Windows Kits\10\Include\10.0.26100.0\um\unknwn.idl
6>64 bit Processing .\KeywordDetectorContosoAdapter.idl
3>64 bit Processing C:\Program Files (x86)\Windows Kits\10\Include\10.0.26100.0\shared\wtypesbase.idl
3>wtypesbase.idl
4>objidl.idl
4>64 bit Processing C:\Program Files (x86)\Windows Kits\10\Include\10.0.26100.0\um\unknwn.idl
1>64 bit Processing C:\Program Files (x86)\Windows Kits\10\Include\10.0.26100.0\shared\basetsd.h
1>basetsd.h
2>unknwn.idl
2>64 bit Processing C:\Program Files (x86)\Windows Kits\10\Include\10.0.26100.0\shared\wtypes.idl
2>wtypes.idl
6>KeywordDetectorContosoAdapter.idl
3>64 bit Processing C:\Program Files (x86)\Windows Kits\10\Include\10.0.26100.0\shared\basetsd.h
2>64 bit Processing C:\Program Files (x86)\Windows Kits\10\Include\10.0.26100.0\shared\wtypesbase.idl
2>wtypesbase.idl
1>64 bit Processing C:\Program Files (x86)\Windows Kits\10\Include\10.0.26100.0\shared\guiddef.h
1>guiddef.h
4>unknwn.idl
4>64 bit Processing C:\Program Files (x86)\Windows Kits\10\Include\10.0.26100.0\shared\wtypes.idl
6>64 bit Processing C:\Program Files (x86)\Windows Kits\10\Include\10.0.26100.0\um\oaidl.idl
6>oaidl.idl
3>basetsd.h
3>64 bit Processing C:\Program Files (x86)\Windows Kits\10\Include\10.0.26100.0\shared\guiddef.h
3>guiddef.h
1>64 bit Processing C:\Program Files (x86)\Windows Kits\10\Include\10.0.26100.0\um\ocidl.idl
[... 377 lines cut ...]
2>Generating Code...
4>   Creating library x64\Debug\DelayAPO.lib and object x64\Debug\DelayAPO.exp
3>SwapAPOSFX.cpp
4>DelayAPO.vcxproj -> C:\Users\vrind\source\repos\Windows-driver-samples\audio\sysvad\APO\DelayAPO\x64\Debug\DelayAPO.dll
7>adapter.cpp
2>   Creating library x64\Debug\KWSApo.lib and object x64\Debug\KWSApo.exp
2>KwsAPO.vcxproj -> C:\Users\vrind\source\repos\Windows-driver-samples\audio\sysvad\APO\KWSApo\x64\Debug\KWSApo.dll
7>basetopo.cpp
7>BthhfpDevice.cpp
7>common.cpp
3>Generating Code...
7>hw.cpp
7>kshelper.cpp
3>   Creating library x64\Debug\SwapAPO.lib and object x64\Debug\SwapAPO.exp
3>SwapAPO.vcxproj -> C:\Users\vrind\source\repos\Windows-driver-samples\audio\sysvad\APO\SwapAPO\x64\Debug\SwapAPO.dll
7>savedata.cpp
7>tonegenerator.cpp
7>UsbHsDevice.cpp
7>hdmitopo.cpp
7>micintopo.cpp
7>spdiftopo.cpp
7>Generating Code...
8>------ Build started: Project: package (Package\package), Configuration: Debug x64 ------
========== Build: 8 succeeded, 0 failed, 0 up-to-date, 0 skipped ==========
========== Build completed at 18:40 and took 12.510 seconds ==========`

forderud, 2024-06-20:
I just tried to build the `audio\sysvad\sysvad.sln` solution in Debug|x64 on the "main" branch and then ended up with the following INF files:  
[image]

There's also a `TabletAudioSample.cer` certificate in the same `audio\sysvad\TabletAudioSample\x64\Debug` subfolder.

vrinda41198, 2024-06-21:
Woah that's weird. I tried to build the same file (on a Windows 11, 23H2 version system) in Debug | x64; but I see a plethora of errors for which I had to update include and library directory properties (to reflect WDK path) and I also had to add/modify a few other settings. In fact I had to manually generate the .idl files (such as DelayAPODll.idl) in between using the corresponding .h files. 

Is there some step in the setup that you did? I just cloned the repository, opened any one of the solutions (like the TabletAudioSample solution) which also opened the remaining 7 solutions, and then built the sysvad solution from solution explorer.

Any input would be really helpful! Thanks!

forderud, 2024-06-21:
Not sure if you've already read the https://github.com/microsoft/Windows-driver-samples/tree/main/audio/sysvad readme, but you need to do the following _before_ building:
* `git submodule update --init` to pull in Windows Implementation Library (WIL)
* Install the `ATL for latest v143 build tools with Spectre Mitigations (x86 & x64)` Visual Studio component.

Everything builds cleanly  for me in VS2022 after following the above steps.

vrinda41198, 2024-06-21:
I did do both of those steps before I built the solution. I even tried to do the setup on another Windows system and ran into the exact same set of errors. This is my first time using Visual Studio or building solution on it. Is there anything I should be doing there that I might have missed?

forderud, 2024-06-24:
> This is my first time using Visual Studio or building solution on it. Is there anything I should be doing there that I might have missed?

You'll need to install Visual Studio with the "Desktop development with C++" workload. Also, you need to install the [WDK](https://learn.microsoft.com/en-us/windows-hardware/drivers/download-the-wdk) for driver development. I'm not aware of any other prerequisites - besides the ATL Spectre one mentioned above.

> Also did you mean v142 build tools with Spectre Mitigations (x86 & x64) instead of v143; since the README mentions the former?

I think that depends on whether you're using Visual Studio 2019 or 2022. When using VS 2022 I needed to install the v143 version.

vrinda41198, 2024-06-25:
[image]
[image]
[image]
[image]
[image]
[image]

I am using a Windows 11 Home 23H2 version OS and I uninstalled everything and re-did the whole process in the same way you mentioned. I am using Visual Studio 2022 Community version, I installed the "Desktop Development with C++" and "Windows Application Development" workloads, as well as all the needed v143 tools (see attached images for installed components), I ran the git submodule command, installed WDK (version 22621.0, according to this [link](https://learn.microsoft.com/en-us/windows-hardware/drivers/other-wdk-downloads) which matches with the SDK version as well.

However, when I built the solution, I again see the same set of errors I saw when I tried to run the solution last time. I am really not sure what I am missing or doing incorrectly.

forderud, 2024-06-25:
I would recommend you to focus on the _first_ error in the build log. The other errors might very well be consequences of the first error.

As for the missing `portcls.h` error, then that header is found on `C:\Program Files (x86)\Windows Kits\10\Include\10.0.22621.0\km\portcls.h` on my computer. You should verify that you've got the header installed. It might be located in a folder with different version than 10.0.22621.0.  Afterwards, verify that the `$(KM_IncludePath)` Visual Studio variable points to the correct folder:  
[image]

If this is not the case, then there's probably a problem with your [WDK](https://learn.microsoft.com/en-us/windows-hardware/drivers/download-the-wdk) installation or WDK Visual Studio extension. You can check if you've installed the latter from Visual Studios "Manage Extensions" menu:  
[image]

Also, you might need to reboot after the WDK installation.

There's **no need for manually changing any project settings** in the samples. Everything should build out-of-the-box if you've installed the required Visual Studio components and WDK.

vrinda41198, 2024-07-03:
I found the issue... it seems that the latest SDK was automatically installed as part of some other installation (maybe VS), and so even though I had installed the right SDK, it was not detecting that.. uninstalling the incorrect SDK helped resolve the issue!

closed 2024-07-03 by vrinda41198