microsoft/Windows-driver-samples #1184: 【 Indirect display 】【IddSampleDriver】 The Idd sample driver has a problem in updating the display mode

N0exp, 2024-06-14:
For example, add two virtual monitors A+B, and then use IddCxMonitorUpdateModes to modify the display mode of the specified monitor, but only A actually changes, and in no case can it change the display mode of B. The first parameter of IddCxMonitorUpdateModes has no real effect. Passing either IDDCX_MONITOR of A or IDDCX_MONITOR of B will change the display mode of A. If I remove monitor A and use IddCxMonitorUpdateModes to change the display mode of B, The parameter error STATUS_INVALID_PARAMETER is returned

N0exp, 2024-08-05:
I execute the above function on win11, can modify the display Settings, but it still fails on win10, how can I modify it to use normally on win10？

Rickylss, 2024-10-24:
Check your win10 version with `winver.exe`. 

**The build number of your win10 version should greater than 19044, if not, do upgrade.**

There may have some bugs in previous win10. But I'm not sure. I did a lot tests before I found this conclusion.