microsoft/Windows-driver-samples #1197: Request: Wi-Fi: WiFiCx reference driver

krish2718, 2024-07-14:
Hi, 

Starting from Windows11, WDI miniport driver is deprecated and WiFiCx should be used, https://learn.microsoft.com/en-us/windows-hardware/drivers/netcx/writing-a-wificx-client-driver but I don't see any reference driver, can you please add to the network/wlan folder?

krish2718, 2025-09-29:
@v-junyli can you please share the commit/PR that introduced this?

Zero3K, 2025-09-29:
There is no such commit/PR.

Zero3K, 2025-09-29:
@krish2718 There is no such commit/PR. Otherwise it would've been mentioned by the system message.

krish2718, 2025-09-29:
So is this not implemented? I saw this marked as completed hence the query.

v-junyli, 2025-09-29:
@krish2718 Hi we're trying to close issues that have been stale for over a year, please feel free to re-open if the issue is still relevant, I reactivated this one.

krish2718, 2025-09-29:
Super thanks, this is still relevant