microsoft/Windows-driver-samples #1206: Fail to run HLK test for Serial device driver

longnhieu7, 2024-08-16:
I am getting in trouble with HLK testing on New Windows11 24H2 about DRIVER_ISOLATION_VIOLATION. I want to create PortName to Registry 'HKEY_LOCAL_MACHINE\HARDWARE\DEVICEMAP\SERIALCOMM’ in pnp.c (My program same as Window driver example on Github for W11 22H2) I caught errors "Registry operations should not use absolute paths. Detected creation of unisolated registry key..." and "Registry operations should only use key handles returned from WDF or WDM APIs. Detected creation of unisolated registry key...".
My INF DIRIDS value is 12, is this point make my driver cannot set value into registry?

longnhieu7, 2024-08-16:
Mr @zlockard told me to test on HLK version 22H2 so I could pass this test. But I would like to make sure that the problem does not come from DIRIDS value and the API we used to set value of COM port. Thank everyone for support!!!

zlockard, 2024-08-20:
Using DIRD 13 is always recommended and won't cause any issues here

closed 2024-08-20 by zlockard