microsoft/Windows-driver-samples #1210: Cannot open wdf.h source file

jahnavidubisetty, 2024-08-29:
Encountering an error saying that "cannot open wdf.h source file".

v-junyli, 2024-10-18:
Hi,

Can you include more info on how to repro the issue? Like what version of Visual Studio and WDK.

saraphinesER, 2024-11-21:
> Encountering an error saying that "cannot open wdf.h source file".

Not sure if this helps since you didn't mention much detail, so this is just for your information.

When I tried to build Toaster project, I also suffer this error "can't open wdf.h",
then I modify the "Windows Driver Kit" checkbox in Visual studio community installer as below.
[image]

I expect this should help, WDK path should be added into system, somehow it's still failed after update, 
also I can't not find wdf.h by manually search "C:\Program Files (x86)\Windows Kits\".
So I download Windows Driver Kit (WDK) from Microsoft, do installation 
and add the path of wdf.h into "Include Directories" of Toaster project.
Then the error is fixed.