microsoft/Windows-driver-samples #1213: Help:  update the FW of USB device through WU

AlbertG123, 2024-09-11:
we have USB device plug in the system,  these USB devices use HID inbox driver.
and these device are following DFU spec （https://www.usb.org/sites/default/files/DFU_1.1.pdf）.
we are going to workout a driver to update the firmware on these USB device. 
can you help guide me which sample code is for this case? 
thank you very much

AndrewYangMSFT, 2024-11-19:
The recommended firmware update solution is CFU, not DFU. Please see this document for more information.

https://learn.microsoft.com/en-us/windows-hardware/drivers/cfu/

closed 2024-11-19 by AndrewYangMSFT

forderud, 2025-04-25:
@AndrewYangMSFT I see that you're recommending CFU for FW updates for HID devices. The https://github.com/microsoft/CFU project with samples and documentation does currently appear to be in a broken and unbuildable state. Multiple issues have been reported the past few years, and there's 9 open pull-requests with proposals for fixing concrete issues.

Would it be possible for someone from Microsoft to review the PRs and help revive the CFU project, so that it becomes easier for HW vendors to learn and adopt CFU?