microsoft/Windows-driver-samples #1214: KASAN example

SuibianP, 2024-09-12:
According to <https://www.microsoft.com/en-us/security/blog/2023/01/26/introducing-kernel-sanitizers-on-microsoft-platforms/>, Kernel AddressSanitizer (KASAN) is available on Windows for kernel-mode. However, there is very limited documentation as to how to use it. It would be nice to have an example detailing the way to build a driver with KASAN support.

PeterGV, 2024-11-04:
In addition to the sample, it would be *extremely* helpful if the documentation described, specifically, how KASAN differs from/adds to the already existing Driver Verifier (which does very efficient pool overrun checking as well as use-after-free checks) and stack overrun checking.
Why would a KM dev want to use KASAN if I already have Verifier enabled?  Can they co-exist? Do they interact? What are the costs/benefits/tradeoffs.

JakobL-MSFT, 2024-11-06:
Huge shoutout to @JohnMcSandwich for making the KASAN scenario work for 3rd party device drivers and even providing a sample.  

To provide a high-level answer to Peter's question:

- DV and KASAN are complimentary.
- DV is "lightweight" (as you do not need to special compile) whereas KASAN requires a special compilation
- DV can only "see" what goes on "on the edge" of the driver whereas KASAN can intercept "inside" of a driver.  (Exception to this: DV does have special treatment of memory manager / IRQL style checks where we bug check based on page violations rather than based on the specific bug in the driver itself (so bugcheck based on a symptom rather than necessarily at the immediate point of failure).  
- DV has tons of driver specific rules.  KASAN is focused on generic memory checks.

I know what you really want is a more comprehensive and data driven answer. I don't have that.

I bet there are many other angles, but that's my 2 cents...

Jakob

References: 
- https://www.microsoft.com/en-us/security/blog/2023/01/26/introducing-kernel-sanitizers-on-microsoft-platforms/
- https://learn.microsoft.com/en-us/windows-hardware/drivers/devtest/kasan
- https://github.com/microsoft/Windows-driver-samples/blob/main/tools/kasan/samples/KasanDemo-WDM/README.md

closed 2024-11-18 by v-junyli