microsoft/Windows-driver-samples #1227: The sample vhidmini install failed。

lie12huo, 2024-10-30:
When I install the KMDF driver in the vhidmini sample through devcon.exe, the following error is reported:

>>>  [Device Install (UpdateDriverForPlugAndPlayDevices) - root\vhidmini]
>>>  Section start 2024/10/30 09:34:36.816
      cmd: devcon.exe  install vhidmini.inf root\vhidmini
     ndv: INF path: C:\WorkStudio\VirtualKM\Windows-driver-samples\hid\vhidmini2\driver\kmdf\x64\Debug\vhidmini\vhidmini.inf
     ndv: Install flags: 0x00000001
     ndv: {Update Device Driver - ROOT\HIDCLASS\0000}
     ndv:      Search options: 0x00000080
     ndv:      Searching single INF 'C:\WorkStudio\VirtualKM\Windows-driver-samples\hid\vhidmini2\driver\kmdf\x64\Debug\vhidmini\vhidmini.inf'
     dvi:      {Build Driver List} 09:34:36.878
     dvi:           Searching for hardware ID(s):
     dvi:                root\vhidmini
     dvi:      {Build Driver List - exit(0x00000000)} 09:34:36.906
!    ndv:      No matching drivers found in single INF
     dvi:      {DIF_SELECTBESTCOMPATDRV} 09:34:36.922
     dvi:           Default installer: Enter 09:34:36.930
     dvi:                {Select Best Driver}
!    dvi:                     Selecting driver failed(0xe0000228)
     dvi:                {Select Best Driver - exit(0xe0000228)}
!    dvi:           Default installer: failed
!    dvi:           Error 0xe0000228: There are no compatible drivers for this device.
     dvi:      {DIF_SELECTBESTCOMPATDRV - exit(0xe0000228)} 09:34:36.980
!    ndv:      Unable to select best compatible driver. Error = 0xe0000228
     ndv:      No drivers found for device.
     ndv: {Update Device Driver - exit(00000103)}
!    ndv: No better matching drivers found for device 'ROOT\HIDCLASS\0000'.
!    ndv: No devices were updated.
<<<  Section end 2024/10/30 09:34:37.026
<<<  [Exit status: FAILURE(0x00000103)]

How can I solve this problem?

v-junyli, 2024-11-07:
@microsoft/coreos-buses-and-wdf

AndrewYangMSFT, 2024-11-08:
> ```
> dvi: Searching for hardware ID(s):
> dvi: root\vhidmini
> dvi: {Build Driver List - exit(0x00000000)} 09:34:36.906
> ! ndv: No matching drivers found in single INF
> ```

Did you change vhidmini.inf? It looks PnP failed to find the device ID "root\vhidmini" in the INF.

closed 2024-11-19 by lie12huo