microsoft/Windows-driver-samples #1238: [general/toaster] Asking help for Building error.

saraphinesER, 2024-11-20:
Hi, I suffer various errors when trying to build toaster project on x64 configuration.
From issues history I knew there're still people also try to build toaster, so please excuse me if my questions are too fundamental.

There're so many errors thus I tried to build individual by project build order.
First I suffer "redefinition;" on notify project as below.
---------
1>C:\Program Files (x86)\Windows Kits\10\Include\10.0.22621.0\um\winioctl.h(39,1): error C2374: 'GUID_DEVINTERFACE_DISK': redefinition; multiple initialization
1>(compiling source file 'notify.c')
1>    C:\Program Files (x86)\Windows Kits\10\Include\10.0.22621.0\um\winioctl.h(39,1):
1>    see declaration of 'GUID_DEVINTERFACE_DISK'
........ (many similar errors)
---------
I marked "#include <winioctl.h>" and it seems passed the compilation.
then next comes error on "No Target Architecture",
---------
>------ Rebuild All started: Project: wdftoastmon, Configuration: Debug x64 ------
1>toastmon.c
1>C:\Program Files (x86)\Windows Kits\10\Include\10.0.22621.0\shared\ntdef.h(197,1): error C1189: #error:  "No Target Architecture"
1>(compiling source file 'toastmon.c')
1>wmi.c
1>C:\Program Files (x86)\Windows Kits\10\Include\10.0.22621.0\shared\ntdef.h(197,1): error C1189: #error:  "No Target Architecture"
1>(compiling source file 'wmi.c')
1>Generating Code...
1>Done building project "wdftoastmon.vcxproj" -- FAILED.
---------
I search and guess there's no x64 definition, so I add "_AMD64_" on C++\preprocessor in properties settings. It passed.
Next comes no "busenumMof.h" under \bus\dynambus  and \bus\StatBus projects
---------
1>------ Rebuild All started: Project: dynambus, Configuration: Debug x64 ------
1>busenum.c
1>C:\MSFT_win_samples_2\general\toaster\toastDrv\kmdf\bus\dynamic\busenum.h(28,10): error C1083: Cannot open include file: 'busenumMof.h': No such file or directory
1>(compiling source file 'busenum.c')
............
---------
It seems somehow there's no busenumMof.h generated. 
I tried to do it manual but still failed..
----------
C:\MSFT_win_samples\general\toaster\toastDrv\kmdf\bus\static>wmimofck busenum.mof

Microsoft (R) WDM Extensions To WMI MOF Checking Utility  Version 1.50.0000
Copyright (c) Microsoft Corp. 1997-2000. All rights reserved.
busenum.mof (0) : error RC2135 : Error accessing binary mof file f, code 0
----------

also others error like
----------
\ms_wdk\e26100\include\wdf\kmdf\1.33\wdfrequest.h(158,52): error C2220: the following warning is treated as an error
1>(compiling source file 'filter.c')
\ms_wdk\e26100\include\wdf\kmdf\1.33\wdfrequest.h(158,52): warning C4324: '<unnamed-tag>': structure was padded due to alignment specifier
1>(compiling source file 'filter.c')
----------

My environment is listed as below, both Visual studio community IDE and MSBuild cmd line can't succeed. 
I really want to know how to build Toaster project correctly, this is just the first step while I stuck.
Please advise if any, any suggestions would be helpful to me.
==============
Environment: 
Visual Studio Community 2202 (64-bit) Version 17.12.0
SDK: Windows 11 SDK (10.0.22621.0)
WDK: Windows driver kit - 10.26100.1
===============

Thanks again for your patience.

v-junyli, 2024-11-20:
@microsoft/coreos-buses-and-wdf can you take a look?

AndrewYangMSFT, 2024-12-10:
Do you see the same issue after following the steps on this document to install the recommended VS/SDK/WDK versions only?
https://learn.microsoft.com/en-us/windows-hardware/drivers/download-the-wdk

After following the recommended steps, without changing any codes, what's the first error? and which project fails building?

saraphinesER, 2024-12-31:
Sorry for late reply and thanks for the guidance. 
I just realize the WDK option in VS installation is something like API, not the real WDK content.
I will download the WDK from the link.
Thanks again.

closed 2024-12-31 by saraphinesER