microsoft/Windows-driver-samples #1239: [RNDIS][Open source][share code]

gg040431204, 2024-11-22:
hi  everyone:
                   I have a question.  Does it have RNDIS open source sample code?  I want to learn this RNDIS driver.  I find such link  https://github.com/microsoft/Windows-driver-samples/tree/main/network/RNDIS  does not exist.
                   So  can you share this sample code to git?

v-junyli, 2024-11-23:
@microsoft/network-driver-platform

gg040431204, 2024-11-25:
hi  v-junyli:
     can you show the detailed link? I can't find it through the key word "network-driver-platform". Thanks

v-junyli, 2024-11-25:
@gg040431204 hi sorry "microsoft/network-driver-platform" is the alias of the owner team of NDIS driver samples, I tagged them to see if they have better information for you about this request.

gg040431204, 2024-11-26:
Hi @v-junyli :
     thanks for your reply. hope to get answer ASAP.

gg040431204, 2024-12-04:
hi  @Shyamal-V :
     Can you share RNDIS sample code? Thanks.

gg040431204, 2024-12-19:
Hi @Shyamal-V :
Can you share RNDIS sample code?Thanks.

gg040431204, 2025-01-08:
Hi @Shyamal-V ：
Can you share RNDIS sample code?Thanks.

Zero3K, 2025-09-02:
No one seems to care.