microsoft/Windows-driver-samples #1242: Does one need a "Certificate" as well to use APO

karl-police, 2024-11-30:
I heard that if you want to create a driver e.g. ``test.sys`` and use it without restriction, it needs to be signed.

But what about APO? These COM things or whatever it is. Do custom APO also have to be signed or not?

Can someone please tell me?

v-junyli, 2024-12-02:
@microsoft/windowsaudio can you please take a look

mvaneerde, 2024-12-03:
Audio processing objects load in audiodg.exe which runs as a child process of the audiosrv service in session 0, and therefore need to be signed. If you're installing on a test system, test-signing is okay. If you're installing a production system they need to be production-signed; Microsoft will production-sign your driver for you if you submit it to Hardware Dev Center.

Test instructions here: https://github.com/microsoft/Windows-driver-samples/tree/main/audio/sysvad

Submission to Hardware Dev Center instructions here: https://learn.microsoft.com/en-us/windows-hardware/drivers/dashboard/hardware-submission-create

More on signing drivers: https://learn.microsoft.com/en-us/windows-hardware/drivers/develop/signing-a-driver

karl-police, 2025-01-11:
> Audio processing objects load in audiodg.exe which runs as a child process of the audiosrv service in session 0, and therefore need to be signed. If you're installing on a test system, test-signing [...]

A working APO ``.DLL`` from somewhere else, doesn't have any License that contains the word "Microsoft", only SHA1, but it runs. The process to sign a driver... not sure, is it free? There's a lot of steps and apparently it asks for things...

mvaneerde, 2025-01-11:
You can use [signtool.exe](https://learn.microsoft.com/en-us/windows/win32/seccrypto/signtool) to verify the signature on the working APO. It might be an embedded signature, or the APO may have an entry in a .cat file which is itself signed

karl-police, 2025-03-03:
> You can use [signtool.exe](https://learn.microsoft.com/en-us/windows/win32/seccrypto/signtool) to verify the signature on the working APO. It might be an embedded signature, or the APO may have an entry in a .cat file which is itself signed

I used ``signtool verify`` on all the ``.DLL`` and the one in System32

They all give

```
SignTool Error: A certificate chain processed, but terminated in a root
        certificate which is not trusted by the trust provider.

Number of errors: 1
```

which is good, that means they work without being uhhh idk?

```
Issued to: AddTrust External CA Root
Issued by: AddTrust External CA Root

Issued to: UTN-USERFirst-Object
Issued by: AddTrust External CA Root

Issued to: COMODO Time Stamping CA
Issued by: UTN-USERFirst-Object

Issued to: Sectigo SHA-1 Time Stamping Signer
Issued by: COMODO Time Stamping CA
```

It is an expired thing though.

Also

```
Issued to: USERTrust RSA Certification Authority
Issued by: USERTrust RSA Certification Authority
```

not sure if this has a specific meaning this "USER" part

mvaneerde, 2025-03-03:
It's important to run the `signtool.exe` command on the machine that you are attempting to load the driver on. `signtool.exe` verifies that the .dll hasn't been tampered with since it was built, and it also verifies that it was built by someone the machine is configured to trust.
As I read the output, the .dll matches the signature, but the signature is by someone that the machine doesn't trust.
See the steps at [Preparing a Computer for Manual Driver Deployment](https://learn.microsoft.com/en-us/windows-hardware/drivers/develop/preparing-a-computer-for-manual-driver-deployment) to get the root certificate added to the right certificate store on your test machine

closed 2025-09-29 by v-junyli