microsoft/Windows-driver-samples #1246: devcon install failed on an arm-based Windows 11

AriesSunMS, 2024-12-13:
Hello
I attempted to deploy sysvad on an ARM-based Windows 11 PC.
However, I am encountering issues with the devcon install ComponentizedAudioSample.inf Root\Sysvad_ComponentizedAudioSample
It always returns a failure, no matter with the debug or release version of the driver

Here is the failure log in Windows\INF. Could anyone help with this issue? Thank you!

>>>  [Device Install (UpdateDriverForPlugAndPlayDevices) - Root\Sysvad_ComponentizedAudioSample]
>>>  Section start 2024/12/13 06:34:41.926
      cmd: E:\\tools\\devcon  install ComponentizedAudioSample.inf Root\Sysvad_ComponentizedAudioSample
     ndv: INF path: E:\x64\Debug\package\ComponentizedAudioSample.inf
     ndv: Install flags: 0x00000001
     ndv: {Update Device Driver - ROOT\MEDIA\0004}
     ndv:      Search options: 0x00000080
     ndv:      Searching single INF 'E:\x64\Debug\package\ComponentizedAudioSample.inf'
     dvi:      {Build Driver List} 06:34:42.051
     dvi:           Searching for hardware ID(s):
     dvi:                root\sysvad_componentizedaudiosample
     dvi:      {Build Driver List - exit(0x00000000)} 06:34:42.082
!    ndv:      No matching drivers found in single INF
     dvi:      {DIF_SELECTBESTCOMPATDRV} 06:34:42.114
     dvi:           Default installer: Enter 06:34:42.129
     dvi:                {Select Best Driver}
!    dvi:                     Selecting driver failed(0xe0000228)
     dvi:                {Select Best Driver - exit(0xe0000228)}
!    dvi:           Default installer: failed
!    dvi:           Error 0xe0000228: There are no compatible drivers for this device.
     dvi:      {DIF_SELECTBESTCOMPATDRV - exit(0xe0000228)} 06:34:42.192
!    ndv:      Unable to select best compatible driver. Error = 0xe0000228
     ndv:      No drivers found for device.
     ndv: {Update Device Driver - exit(00000103)}
!    ndv: No better matching drivers found for device 'ROOT\MEDIA\0004'.
     ndv: {Update Device Driver - ROOT\MEDIA\0005}
     ndv:      Search options: 0x00000080
     ndv:      Searching single INF 'E:\x64\Debug\package\ComponentizedAudioSample.inf'
     dvi:      {Build Driver List} 06:34:42.270
     dvi:           Searching for hardware ID(s):
     dvi:                root\sysvad_componentizedaudiosample
     dvi:      {Build Driver List - exit(0x00000000)} 06:34:42.317
!    ndv:      No matching drivers found in single INF
     dvi:      {DIF_SELECTBESTCOMPATDRV} 06:34:42.332
     dvi:           Default installer: Enter 06:34:42.348
     dvi:                {Select Best Driver}
!    dvi:                     Selecting driver failed(0xe0000228)
     dvi:                {Select Best Driver - exit(0xe0000228)}
!    dvi:           Default installer: failed
!    dvi:           Error 0xe0000228: There are no compatible drivers for this device.
     dvi:      {DIF_SELECTBESTCOMPATDRV - exit(0xe0000228)} 06:34:42.410
!    ndv:      Unable to select best compatible driver. Error = 0xe0000228
     ndv:      No drivers found for device.
     ndv: {Update Device Driver - exit(00000103)}
!    ndv: No better matching drivers found for device 'ROOT\MEDIA\0005'.
     ndv: {Update Device Driver - ROOT\MEDIA\0006}
     ndv:      Search options: 0x00000080
     ndv:      Searching single INF 'E:\x64\Debug\package\ComponentizedAudioSample.inf'
     dvi:      {Build Driver List} 06:34:42.489
     dvi:           Searching for hardware ID(s):
     dvi:                root\sysvad_componentizedaudiosample
     dvi:      {Build Driver List - exit(0x00000000)} 06:34:42.520
!    ndv:      No matching drivers found in single INF
     dvi:      {DIF_SELECTBESTCOMPATDRV} 06:34:42.551
     dvi:           Default installer: Enter 06:34:42.567
     dvi:                {Select Best Driver}
!    dvi:                     Selecting driver failed(0xe0000228)
     dvi:                {Select Best Driver - exit(0xe0000228)}
!    dvi:           Default installer: failed
!    dvi:           Error 0xe0000228: There are no compatible drivers for this device.
     dvi:      {DIF_SELECTBESTCOMPATDRV - exit(0xe0000228)} 06:34:42.629
!    ndv:      Unable to select best compatible driver. Error = 0xe0000228
     ndv:      No drivers found for device.
[... 100 lines cut ...]
     ndv: {Update Device Driver - exit(00000103)}
!    ndv: No better matching drivers found for device 'ROOT\MEDIA\0011'.
     ndv: {Update Device Driver - ROOT\MEDIA\0012}
     ndv:      Search options: 0x00000080
     ndv:      Searching single INF 'E:\x64\Debug\package\ComponentizedAudioSample.inf'
     dvi:      {Build Driver List} 06:34:43.817
     dvi:           Searching for hardware ID(s):
     dvi:                root\sysvad_componentizedaudiosample
     dvi:      {Build Driver List - exit(0x00000000)} 06:34:43.848
!    ndv:      No matching drivers found in single INF
     dvi:      {DIF_SELECTBESTCOMPATDRV} 06:34:43.879
     dvi:           Default installer: Enter 06:34:43.879
     dvi:                {Select Best Driver}
!    dvi:                     Selecting driver failed(0xe0000228)
     dvi:                {Select Best Driver - exit(0xe0000228)}
!    dvi:           Default installer: failed
!    dvi:           Error 0xe0000228: There are no compatible drivers for this device.
     dvi:      {DIF_SELECTBESTCOMPATDRV - exit(0xe0000228)} 06:34:43.942
!    ndv:      Unable to select best compatible driver. Error = 0xe0000228
     ndv:      No drivers found for device.
     ndv: {Update Device Driver - exit(00000103)}
!    ndv: No better matching drivers found for device 'ROOT\MEDIA\0012'.
!    ndv: No devices were updated.
<<<  Section end 2024/12/13 06:34:44.004
<<<  [Exit status: FAILURE(0x00000103)]

v-junyli, 2024-12-18:
@microsoft/windowsaudio

mvaneerde, 2024-12-18:
It looks like you built the driver for the Intel/AMD 64-bit instruction set and you're trying to install it on a machine that uses the ARM 64-bit instruction set. This won't work.

Build the driver for the ARM 64-bit instruction set.

The [DDInstall section](https://learn.microsoft.com/en-us/windows-hardware/drivers/install/inf-ddinstall-section) in the built .inf should have a name that contains `.ntarm64` - if it contains `.ntamd64` instead then you built for the wrong instruction set

closed 2024-12-18 by mvaneerde