microsoft/Windows-driver-samples #1260: Cant make the kbfiltr example working

massimilianoomero, 2025-01-20:
This is maybe more a question than an issue.
I'm probably doing something wrong but I can't make the kbfiltr example working.
Driver is not installing properly,
Here's the error:

**This device cannot start. (Code 10)
Insufficient system resources exist to complete the API.**

And his is the log in the driver events
**Device ROOT\DEVGEN\{14BD7B21-F5A0-5E40-8BE8-FE289A97CC87} had a problem starting.
Driver Name: oem156.inf
Class Guid: {4d36e96b-e325-11ce-bfc1-08002be10318}
Service: i8042prt
Lower Filters: 
Upper Filters: kbfiltr
Problem: 0x0
Problem Status: 0xC00000E5**

The only change I did (except some traces I added) was inserting the real hardware id of the device I want to filter.
The device is a footswitch with three pedals simulating a keyboard.
Here the logs in DbgView
--- FOOTSWITCH --- Keyboard Filter Driver Sample - Driver Framework Edition.
--- FOOTSWITCH --- Built Jan 20 2025 19:35:11
--- FOOTSWITCH --- Enter FilterEvtDeviceAdd 
--- FOOTSWITCH --- Entered KbFiltr_CreateRawPdo
--- FOOTSWITCH --- KbFiltr_CreatePdo success 0
--- FOOTSWITCH --- Entered KbFilter_EvtIoInternalDeviceControl
--- FOOTSWITCH --- Entered KbFilter_EvtIoInternalDeviceControl
--- FOOTSWITCH --- hook keyboard received!
--- FOOTSWITCH ---  NT_ASSERT 48 48 - **This is OK !**
--- FOOTSWITCH --- Exiting KbFilter_EvtIoInternalDeviceControl with status 0
--- FOOTSWITCH --- Exiting KbFilter_EvtIoInternalDeviceControl with status 0

Everything seems fine but the driver is not working. I can't understand why.
I'm not new to windows drivers but I barely worked with filter drivers.
I'm not using a coinstaller since I don't know how to use it and I read it's not strictly necessary in Windows 11.

Can someone please help ?

v-junyli, 2025-01-23:
@microsoft/coreos-buses-and-wdf can you please take a look

AndrewYangMSFT, 2025-01-23:
How do you insert the hardware ID? If it's done in your INF, then what were the code changes when you started to see the device start failure for the first time? Maybe you can try to narrow it down by starting over and making changes gradually.