microsoft/Windows-driver-samples #1266: Windows-driver-samples/tree/main/network/trans/stmedit

jdaruja, 2025-01-31:
I could build stmedit driver in VS 2022, and deployed to target computer. But it failed to start 

****Here are the install details**
**C:\Windows\System32>sc.exe create stmedit binpath= c:\windows\system32\stmedit.sys
[SC] CreateService SUCCESS

C:\Windows\System32>sc start stmedit
[SC] StartService FAILED 2:

The system cannot find the file specified.****

Not sure why its failed, any help appreciated

Here is my .INF file 

;;;
;;; Copyright (c) Microsoft Corporation. All rights reserved
;;;
;;; Abstract:
;;;    Stream Edit Callout sample driver install configuration.
;;;

[Version]
    Signature   = "$Windows NT$"
    Class       = WFPCALLOUTS
    ClassGuid   = {57465043-616C-6C6F-7574-5F636C617373}
    Provider    = %Contoso%
    CatalogFile = StmEdit.cat
    DriverVer = 01/31/2025,11.26.27.665
    PnpLockdown = 1

[SourceDisksNames]
   1 = %StmEditDisk%,,,""

[SourceDisksFiles]
   StmEdit.sys = 1,,

[DestinationDirs]
    StmEdit.DriverFiles = 12                              ; %WinDir%\System32\Drivers

[DefaultInstall.NTamd64]
    OptionDesc = %StmEditServiceDesc%
    CopyFiles  = StmEdit.DriverFiles

[DefaultInstall.NTamd64.Services]
    AddService = %StmEditServiceName%,,StmEdit.Service

[StmEdit.DriverFiles]
    StmEdit.sys,,,0x00000040                              ; COPYFLG_OVERWRITE_OLDER_ONLY

[StmEdit.Service]
    DisplayName   = %StmEditServiceName%
    Description   = %StmEditServiceDesc%
    ServiceType   = 1                                                      ; SERVICE_KERNEL_DRIVER
    StartType     = 3                                                      ; SERVICE_DEMAND_START
    ErrorControl  = 1                                                      ; SERVICE_ERROR_NORMAL
    ServiceBinary = %12%\StmEdit.sys                                       ; %WinDir%\System32\Drivers\StmEdit.sys
    AddReg        = StmEdit.AddRegistry

[StmEdit.AddRegistry]
    HKR,"Parameters","InspectionLocalPort",0x00010001,"8888"               ; FLG_ADDREG_TYPE_DWORD
    HKR,"Parameters","InspectionRemotePort",0x00000000,"0"                 ; FLG_ADDREG_TYPE_SZ

[Strings]
    Contoso            = "Contoso Ltd."
    StmEditDisk        = "Stream Edit Installation Disk"
    StmEditServiceDesc = "Stream Edit Callout Driver"
    StmEditServiceName = "StmEdit"

v-junyli, 2025-02-04:
@microsoft/netsec hi can anyone help take a look?

jdaruja, 2025-04-04:
Any help?

SinghNanak, 2025-05-22:
Same here, have you found a solution