microsoft/Windows-driver-samples #1267: Regarding the UDFS driver used by Windows 11

Zero3K, 2025-01-31:
It would be nice if its source code could be made available so that way it can be improved upon by other people.

Zero3K, 2025-01-31:
@AndrewYangMSFT @Shyamal-V 

Also, this issue should have the request label associated to it and I wonder who else I can mention in order to get its source open (it's a pain to find someone to work on DVD Write Now's UDFS driver and then for them to work on it).

Zero3K, 2025-02-01:
@QuintenJohnson @barrygolden @IrinVoso Can any of you help with getting it done?

NeoAdonis, 2025-02-04:
@JakobL-MSFT Can you take a look at this request, please?

Zero3K, 2025-02-04:
Thanks for mentioning him. Now, what about adding the request label to this issue?

Zero3K, 2025-02-07:
@prashantchahar Can you take a look at this request, please?

Zero3K, 2025-02-12:
https://aka.ms/AAud8xs

Zero3K, 2025-02-23:
Anyone?

Zero3K, 2025-02-28:
@dl2n

Zero3K, 2025-03-01:
@darwin-msft

Zero3K, 2025-03-01:
@zlockard

Zero3K, 2025-03-01:
@microsoft/filesystems

v-junyli, 2025-03-05:
@darwin-msft it's filesys related, can you help take a look?

Zero3K, 2025-03-05:
I already tried mentioning him earlier.

closed 2025-03-07 by JakobL-MSFT

Zero3K, 2025-03-07:
Completed? How is it completed? The driver isn't showing up.

Zero3K, 2025-03-07:
I guess the answer to my request is a no.

Zero3K, 2025-03-07:
Is it because I mentioned too many people?