microsoft/Windows-driver-samples #1269: The hid device obtains the path and port

1369841815, 2025-03-03:
Hello, I have a hid Device, and when I look at the source code and click the device, the Device Path will not be displayed, but the tool I use has this path. Now I can get the port of the hub where the device is located, but I cannot make this port correspond to the path of the hid device

Zero3K, 2025-03-04:
What driver are you writing about?

v-junyli, 2025-03-05:
@microsoft/coreos-buses-and-wdf can you please take a look?

AndrewYangMSFT, 2025-03-05:
I'm not sure I understand the problem and your ultimate goal.

If you want to view the device topology, you may use Device Manager and switch to the "Device by connection" view. To check which 
 port it is connected, you may check the "Location" property on the device Properties page. Alternatively, you may use USBVIEW tool.

"Device Instance Path" does not have port number. In general, your codes for HID devices should not be specific to the USB port, because it can be connected to any port.

One thing to note is that if it's a composite device, the HID device is a child device of the USB composite device which is the one connected to the USB port. This could potentially cause a bit confusion.

If more assistance is needed, maybe a screenshot would be helpful.

1369841815, 2025-03-07:
I want to get the location info in the device information on our software to get the physical port linked to the device, but I don't see how to get this information in the routine

AndrewYangMSFT, 2025-03-07:
You may get it by querying this device property of the USB device.
https://learn.microsoft.com/en-us/windows-hardware/drivers/install/devpkey-device-locationinfo

But again, please do not have your software implementation rely on the location, because it can be changed when the end-user changes the port, or when a different system has a slightly different topology.

closed 2025-05-19 by 1369841815