microsoft/Windows-driver-samples #1271: build avcamera project error.

megahertz0, 2025-03-12:
When I build Windows-driver-samples\avstream\avscamera project, It show errors, How can I fix it? 
Windows-driver-samples\avstream\avscamera\sys\avscamera.inx(37-37): error 1318: Included INF "kscaptur.inf" is not part of the target OS platform.
Windows-driver-samples\avstream\avscamera\sys\avscamera.inx(37-37): error 1317: Needed undecorated section [KSCAPTUR.Registration.NT] is not part of the target OS platform.

 Thanks!

v-junyli, 2025-03-17:
@microsoft/cameraplatdev

gerritsMSFT, 2025-03-17:
@megahertz0  What machine are you building on? KSCaptur.inf is included along every Windows installation, and should be found in C:\Windows\INF\KsCaptur.inf. It's extremely odd, since that is also the location for ks.inf.