microsoft/Windows-driver-samples #1288: How to install moufiltr in Window11 with VM PB18?

cer-cer, 2025-04-21:
Hi, i am a newber of filter driver development, i have tried serval times to install the moufiltr in Win11 with VM PB 18, but it alway said the newest driver already installed, which cause the moufiltr can't install.

Could anyone help me of it? thanks.

v-junyli, 2025-04-26:
@microsoft/filter-manager @microsoft/filesystems can anyone help with this?

cgallred, 2025-04-26:
@v-junyli I don't know what "moufiltr" is. This doesn't appear to be a file system filter request.

cgallred, 2025-04-26:
@v-junyli Turns out it is a mouse input filter: https://github.com/microsoft/Windows-driver-samples/tree/main/input/moufiltr

Please reassign to the correct team.

v-junyli, 2025-05-01:
@microsoft/winiotdev hi are you the owner of this area? Looks like it's a mouse related sample.

v-junyli, 2025-05-21:
@microsoft/coreos-buses-and-wdf Hi can you take a look?