microsoft/Windows-driver-samples #1289: sysvad install report error code 39

865972056, 2025-05-16:
[image]

Ask everyone, have you ever encountered it, how to solve it？

v-junyli, 2025-05-21:
have you tried to a clean reinstall

CC @microsoft/windowsaudio

865972056, 2025-05-21:
yes, i have try it; later change win11 is ok, has problem is win 1909

v-junyli, 2025-05-21:
ok good to hear it's solved.

Closing as it does not repro in the latest Windows 11.

closed 2025-05-21 by v-junyli

865972056, 2025-05-23:
Dear author, I have a question to ask you. First of all, thank you for your answer to the previous question
The questions are as follows：
1、In the savedata.cpp file, there is an interface dump to play the data
2、In the tonegenerator.cpp file, is it a recording data dump ？
3、windows c++ How does a programming application need to obtain audio data and deliver recording data？
thinks.