microsoft/Windows-driver-samples #1322: Immediate return mode does not complete ReadFile request as expected (timeout = 0)

Koji-Okuda, 2025-08-27:
### Summary
In the configuration where `ReadIntervalTimeout = MAXULONG`, `ReadTotalTimeoutConstant = 0`, and `ReadTotalTimeoutMultiplier = 0`, the driver is expected to return immediately with available data—even if zero bytes.

However, the current implementation sets `returnWithWhatsPresent = TRUE` and updates `reqContext->Status`, but does not call `WdfRequestComplete`, resulting in no response to the user-mode caller.

### Expected Behavior
The request should be completed immediately via `WdfRequestComplete`, without relying on interrupts or timers.

### Actual Behavior
The request remains uncompleted, causing the user-mode caller to hang.

### Suggested Fix
Explicitly complete the request when `returnWithWhatsPresent = TRUE`, for example:

```c
if (returnWithWhatsPresent) {
    reqContext->Status = STATUS_SUCCESS;
    WdfRequestComplete(request, STATUS_SUCCESS);
    return;
}

This ensures that the driver adheres to the expected semantics of immediate return and maintains consistency across different timeout configurations. It also improves maintainability and enables template-based reuse in other serial or communication drivers.

If my understanding of the timeout semantics or the expected behavior is incorrect, I would greatly appreciate clarification.

v-junyli, 2025-09-05:
@microsoft/coreos-buses-and-wdf

AndrewYangMSFT, 2025-09-05:
Could you please confirm which sample source file and function you are referring to? Is it function SerialStartRead in this file?

https://github.com/microsoft/Windows-driver-samples/blob/main/serial/serial/read.c

Koji-Okuda, 2025-09-06:
After reviewing the sample more carefully, I realized my initial assumption was incorrect.

I had expected the ReadFile request to be completed immediately when timeout values were set to zero, but I overlooked the fact that the sample uses SerialStartRead to synchronously invoke SerialGetNextRequest. This means the request is not deferred to an interrupt or DPC, but is processed immediately if data is available.

My misunderstanding stemmed from assuming that ReadFile completion was entirely event-driven, when in fact the design allows for immediate completion via SerialStartRead.

I apologize for the confusion caused by the issue title and description. I appreciate the clarity this sample provides once the flow is properly understood, and I hope this correction helps others reviewing the code.

Koji-Okuda, 2025-09-06:
I am currently working on a UART driver for the i.MX8MP platform, using this sample as a reference.

If I encounter any issues related to send/receive processing on the i.MX8MP UART, I will post them here and would greatly appreciate any advice or insights.

AndrewYangMSFT, 2025-09-10:
Thank you for the update. Let's close out this one. Please post a new issue when you encounter any later.